frap129/OpenQ1 #2: mcu_configs folder empty 24.2.0-20240508-2203

53Aries, 2024-06-03:
[image]
cp: cannot stat '/home/mks/openq1/mcu_configs/.config.mainboard': No such file or directory

53Aries, 2024-06-03:
Also I cant figure out how to add the files to the klipper/config folder properly. I add them but they dont seem to have any effect on the menuconfig. Trying to decipher the file and select the correct settings manually. Can you just provide a screenshot of the menuconfig settings?

Peck07, 2024-06-03:
The folder is not empty, windows probably doesn't show the .xxxx files.
cp fails because the path is wrong, instead openq1 try OpenQ1 (linux is case sensitive)

frap129, 2024-06-03:
> cp: cannot stat '/home/mks/openq1/mcu_configs/.config.mainboard': No such file or directory
Where is this error from?

> Also I cant figure out how to add the files to the klipper/config folder properly. I add them but they dont seem to have any effect on the menuconfig. Trying to decipher the file and select the correct settings manually. Can you just provide a screenshot of the menuconfig settings?

You're missing the `.` in front of config. They do not go in the config folder, the need to be copied to the klipper folder with the name `.config`.

53Aries, 2024-06-03:
> > cp: cannot stat '/home/mks/openq1/mcu_configs/.config.mainboard': No such file or directory
> 
> Where is this error from?
> 
> > Also I cant figure out how to add the files to the klipper/config folder properly. I add them but they dont seem to have any effect on the menuconfig. Trying to decipher the file and select the correct settings manually. Can you just provide a screenshot of the menuconfig settings?
> 
> You're missing the `.` in front of config. They do not go in the config folder, the need to be copied to the klipper folder with the name `.config`.
> 
> EDIT: What program are you using for SSH?

Putty. I ended up getting it flashed using the menuconfig from https://github.com/leadustin/QIDI-up2date-english. Using rp2040, 16k bootloader, usbserial communication, Flashing was successful but upon reinstalling the toolhead board to the qidi wiring the host no longer boots. If i unplug the toolhead the host will boot. I rechecked my soldering several times to make sure i didn't have two terminals bridge or something. So either i somehow broke the toolhead board or the menuconfig i used is wrong.

frap129, 2024-06-03:
> Putty. I ended up getting it flashed using the menuconfig from https://github.com/leadustin/QIDI-up2date-english. Using rp2040, 16k bootloader, usbserial communication, Flashing was successful but [...]

@Peck07 reported this in the discord earlier today, it doesn't seem to be related to the config. I will be doing a full reinstall on my own printer to investigate this later today.

53Aries, 2024-06-03:
> @Peck07 reported this in the discord earlier today, it doesn't seem to be related to the config. I will be doing a full reinstall on my own printer to investigate this later today.

What discord server? Thanks

frap129, 2024-06-03:
@53Aries [the unofficial QIDI discord](https://discord.com/invite/bzPZmkmr). We have a thread going for custom firmware work in the q-series-forum channel

frap129, 2024-06-05:
Update: This ended up being a klipper/katapult configuration error. Fixed in recent commits.

closed 2024-06-05 by frap129