frap129/OpenQ1 #3: inconsistent z-offset

kripobs, 2024-06-07:
Hello,
first of all, thank you very much for this project.
So far, there have been hardly any problems with this firmware.
Unfortunately, I couldn't get the display to work, but what's worse is that I can't get a reliable first layer. Sometimes it's too flat, sometimes it's somewhere in the middle, but never the same, and sometimes it's too high and the print has to be aborted. Did I miss something when setting it up?

frap129, 2024-06-07:
This is a known issue, and something I plan on addressing soon. For me, this issue is usually solved by letting the bed preheat for a while, but @Peck07 has reported that solution does not work for them. I suggest giving that a try in the meantime.

Some explanation on what I think the problem is: In the stock firmware, QIDI gets the Z-Offset while everything is cold, and applies a fixed 0.1mm offset to the measured value. In OpenQ1, we get the Z-Offset while the bed is hot to account for thermal expansion, then apply a fixed 0.05mm offset. I'll revert this to the original logic soon, once I get time to do some testing. 

This commit should better replicate the stock functionality. [https://github.com/frap129/OpenQ1/commit/2c07ecfd6e67a9e02136bf24734c75d6f289d14c](https://github.com/frap129/OpenQ1/commit/2c07ecfd6e67a9e02136bf24734c75d6f289d14cl)
 I have not tested it, but feel free to give it a shot and see if it fixes your issue.
 
 The screen is only partially functional so far and is not included in the default installation yet. If you would like to manually set it up: 
1. `git clone https://github.com/frap129/klipmi ~/klipmi`
2. `cd ~/klipmi`
3. `./install.sh`
4. `systemctl stop klipmi`
5. `.venv/bin/nextion-fw-upload -b 115200 -ub 921600 /dev/ttyS1 ~/klipmi/hmi/openq1/Q1_UI.tft`
6. `~/moonraker/scripts/fetch-apikey.sh` (save the output for the next step)
7. Write the following in ~/printer_data/config/klipmi.toml
```
[klipmi]
device = "/dev/ttyS1"
baudrate = 115200
ui = "openq1"

[moonraker]
host = "0.0.0.0"
port = 7125
api-key = "Api key from previous step"
```
8. `systemctl start klipmi`

kripobs, 2024-06-07:
Thank you very much for the quick reply. I have tried with probing with pre heating couldn't detect any less deviation. some times it measures ~2mm sometimes only ~1.5mm. but never the perfect middle (hot or cold).
i wonder why the qidi way works so good (i found the first layer on the stock firmware really good), considering the fact that they use such an unprecise method.
thank you again for the display guide. the process crashes consistently at 40%. i will investigate that.

frap129, 2024-06-17:
The Z offset issues should be resolved with the latest updates to OpenQ1 and qidi_auto_z_offset. Please follow the calibration steps in [https://github.com/frap129/qidi_auto_z_offset?tab=readme-ov-file#basic-usage](https://github.com/frap129/qidi_auto_z_offset?tab=readme-ov-file#basic-usage)

closed 2024-06-17 by frap129