frap129/OpenQ1 #6: Orca Slicer

Pukkahq, 2024-06-17:
I found this project the other day and have ordered an EMMC card so I can flash my Q1 Pro.

Do I need to change any start or end macros in orca slice to OpenQ1?

If I want to revert back to the stock firmware is this a straight forward process?

frap129, 2024-06-17:
You do not need to change your start/end macros in OrcaSlicer. In the layer change gcode in OrcaSlicer, it calls "LOG_Z", a macro that does not exist in OpenQ1. You do not need to remove this, but your klipper console will show "Unknown command LOG_Z".

Reverting back is easy, all that needs to be done is building/flashing klipper for the MCU's using the stock EMMC card. I do plan on writing a "Getting back to stock" guide at some point, though the process is almost the same as installing OpenQ1

Pukkahq, 2024-06-17:
Thanks for the quick reply 😄 

Just waiting for the postman to arrive now!

closed 2024-06-18 by frap129

vonwerderc, 2024-08-03:
> Thanks for the quick reply 😄
> 
> Just waiting for the postman to arrive now!

Did you get this installed? Curious how easy it is to get up and running. I'm not incredibly confident in my soldering ability.