frap129/OpenQ1 #15: Filament Runout does not trigger

Ivanuke, 2024-08-26:
The runout sensor does not get triggered. The tangle sensor works as expected and resumes as normal. Is the runout sensor configured with the Open Q1 software?

damanbaird, 2024-08-31:
Dou you have this: 

pause_on_runout:True
runout_gcode:
            RESET_FILAMENT_WIDTH_SENSOR
            M118 Filament run out

Ivanuke, 2024-09-02:
Yes I have that in the q1.cfg file

frap129, 2024-09-07:
Sorry for the delay. I also had this issue once, I believe this is related to [QIDI_Q1_Pro/issues/65](https://github.com/QIDITECH/QIDI_Q1_Pro/issues/65). I can try to test using the values from that issue, but the values in open q1 have also been calibrated separately. Ultimately, the sensor itself is garbage, and there may not be much that can be done to solve it.

konstantin-kelemen, 2024-11-05:
I did have the same issue. I solved it by changing the following value in the `[hall_filament_width_sensor]` section of `q1.cfg`:
`min_diameter: 1.0`
It works fine for my sensor and didn't give any false positives yet.

Ivanuke, 2024-11-05:
Thank you I will give this a try!

frap129, 2024-11-06:
I'll try it out as well!

konstantin-kelemen, 2024-11-06:
To make sure that the value is not too high or too low for your particular machine, remove any filament from the sensor and run `query_filament_width` in the fluidd console. Mine gave me a value of about 0.89mm. This means that this would never go lower that that and a default value of 0.3 will never trigger the sensor.

konstantin-kelemen, 2024-11-11:
Has anybody tried the proposed fix yet? It would be nice to implement it in the repo so I don't have to change the code on my printer after each update 😄

frap129, 2024-11-12:
> Has anybody tried the proposed fix yet? It would be nice to implement it in the repo so I don't have to change the code on my printer after each update 😄

I haven't finished a roll yet, not much printing lately. I'll take Ivanuke's word and make the change. For future reference though, you can put overriding changes in your printer.cfg so that you don't have to reapply the changes every update. This change, for example, just add 
```
[hall_filament_width_sensor]
min_diameter: 1.0
```
anywhere below the include statements in your printer.cfg, and it will not interfere with updates.

frap129, 2024-11-12:
@Ivanuke Can we consider this issue solved?

frap129, 2024-11-12:
Fixed with commit d4344897fa45df95147600e3d9b587e05a870498. Thanks again @konstantin-kelemen

closed 2024-11-12 by frap129