frap129/OpenQ1 #16: Auto Z-Offset Calibration problem

MicroSur, 2024-09-24:
Hello!

Can you give more detailed instructions on Auto Z-Offset Calibration?

3. Run AUTO_Z_CALIBRATE (OK)
4. Move Z to 0 ( Which way? G1 Z0 or use manual z-offset adjustment from the web interface menu? )
5. Verify that you can slide a piece of paper under the nozzle ( there is no gap, should I use manual z-offset adjustment from the web interface menu? )

I ask because I can't get a satisfactory result, the nozzle is too low when printing.
Thanks.

frap129, 2024-10-01:
If you can't slide a piece of paper under the nozzle, adjust the z offset until you can

tschreiner, 2024-11-08:
@MicroSur First run G1 Z0. It should go to Z=0 with the offset applied (so in reality like -1.9mm or so). Then adjust the offset manually and run AUTO_Z_SAVE_GCODE_OFFSET and SAVE_CONFIG Now it should be fine.

closed 2024-11-12 by frap129