frap129/OpenQ1 #19: Question regarding Z offset

szafranski, 2024-11-12:
Hi! Awesome initiative! Just a quick question. Is Z offset being calibrated automatically or not? There are 2 informations - one that states that is it not supported and one that it is. Thanks!

frap129, 2024-11-12:
It is measured automatically (no paper test or other manual setup), but you have to run the command manually. See https://github.com/frap129/OpenQ1/blob/main/docs/Calibrating-Z-Offset.md

szafranski, 2024-11-12:
> It is measured automatically (no paper test or testing), but you have to
> run the command manually. See
> https://github.com/frap129/OpenQ1/blob/main/docs/Calibrating-Z-Offset.md

In the link you sent there is exact description of paper test in point 5

frap129, 2024-11-12:
Ah I see the confusion. When I said paper test, I mean the standard 3d printer "put a piece of paper under the nozzle, and repeatedly adjust the nozzle by 0.05mm until the resistance feels right", whereas this module only asks you to make sure the nozzle isn't physically touching the bed.

frap129, 2024-11-12:
I should note, that step only exists because some people have adjusted the bed screws on their Q1, which breaks the auto-calibrating system. The sensors are attached through the bed screws and require a specific amount of pressure to be accurate.

aidan-gibson, 2024-11-13:
I'm also having issues with this. I click home all (manually inputting XYZ values in fluidd doesn't work for me), then type `AUTO_Z_CALIBRATE` and get `!! Probe triggered prior to movement` every time.

aidan-gibson, 2024-11-14:
Thanks, I'll try that out! I haven't adjusted the bed screws at all.

closed 2025-01-07 by frap129