frap129/OpenQ1 #22: Note: Camera

aidan-gibson, 2024-11-13:
By the way, I had to manually install `ustreamer` to get the camera to work.

aidan-gibson, 2024-11-13:
ALSO: changing crowsnest.conf line 45 to 30 max fps would be advisable: unsure how much QIDI has changed the Q1 during the time its been out, but mine supports 30fps at 1920x1080 (MJPG). When I manually make changes to the files that are "locked" in fluidd it breaks updates from y'all, it would suck to unapply and reapply this setting on every update.

frap129, 2024-11-14:
Thanks for finding that. `ustreamer` should be installed in the base image, so I'll have to investigate a bit. As for the FPS, the built in WiFi dongle sucks. I've had many issues with the WebUI disconnecting if the framerate is set too high. Are you wired in by ethernet?

frap129, 2024-11-15:
> Still, you can limit it to 15 fps if you so desire from the fluidd UI if I'm not mistaken?

Then everyone using the wifi dongle on open Q1 would have to make that change to prevent bandwidth issues. `~/printer_data/config/crowsnest.conf` is just a link to `~/OpenQ1/configs/crowsnest.conf` by default. You can run `rm ~/printer_data/configs/crowsnest.conf` and `cp ~/OpenQ1/configs/crowsnest.conf ~/printer_data/configs/crowsnest.conf` to un-link from the OpenQ1 version, then edit it all you want. You'll still get updates from OpenQ1, just not for that specific file.

frap129, 2024-11-15:
> By the way, I had to manually install `ustreamer` to get the camera to work.

Upon further investigation, this doesn't make sense. Crowsnest bundles its own modified version of ustreamer. What other steps did you take to solve it? Rebooting the machine, restarting crowsnest, etc

aidan-gibson, 2024-11-25:
> Then everyone using the wifi dongle on OpenQ1 would have to make that change to prevent bandwidth issues. `~/printer_data/config/crowsnest.conf` is just a link to `~/OpenQ1/configs/crowsnest.conf` [...]

THANK YOU, I should've dug and figured this out myself but this is exactly what I needed to know, appreciate it.

aidan-gibson, 2024-11-25:
> this doesn't make sense

`apt` moment fr! I wouldn't worry about it then, maybe I did something weird, I don't recall at this point. But `ustreamer` definitely wasn't there for some reason. Closing issue.

closed 2024-11-25 by aidan-gibson