frap129/OpenQ1 #25: Stepper motors not disabled after print

deveth0, 2024-11-24:
In contrary to the stock firmware, OpenQ1 does currently not disable the stepper motors after printing. That results in a power usage of around 50W idle (compared to 14W) and rather high temperatures.

There is already a `idle_timeout` block but somehow it does not trigger?

frap129, 2024-11-24:
Do you have any updates available for OpenQ1 under the Web UI? This should have been fixed by the most recent commit, which caused idle_timeout to not disable motors due to a typo

deveth0, 2024-11-24:
already have this change in my config.

frap129, 2024-11-24:
I have just pushed a commit (e5ab9b9fe3f7369cd8ac05b788627ddeb87274d1) that restores the stock M84 functionality, which leaves the Z steppers enabled. It is possible that the motors have an inverted enable/disable pin, causing them to remain enabled even after M84.

deveth0, 2024-11-24:
Will give it a try, although disabling the steppers in Fluidd works fine using M84

deveth0, 2024-12-01:
Sorry, that did not help

Peck07, 2024-12-28:
@deveth0 can you point out which steppers are still enabled exactly?

deveth0, 2024-12-28:
All of em

Peck07, 2024-12-29:
Just to be clear, even after issuing an M84? Tried it yesterday, and at least I could move the head freely.
Should be something else, like frap129 mentioned.

deveth0, 2024-12-31:
I now replaced the `PRINT_END` macro with the original one and that works fine

```
[gcode_macro PRINT_END]
gcode:
    SET_IDLE_TIMEOUT TIMEOUT={printer.configfile.settings.idle_timeout.timeout}         ; set timeout back to configured value
    CLEAR_PAUSE
    #save_zoffset
    M106 P2 S0
    M106 P0 S0
    M106 P3 S0
    
    M104 S0
    M140 S0
    M141 S0

    M220 S100
    M221 S100
    SET_STEPPER_ENABLE STEPPER=stepper_x enable=0
	SET_STEPPER_ENABLE STEPPER=stepper_y enable=0
	SET_STEPPER_ENABLE STEPPER=stepper_z enable=1
    SET_STEPPER_ENABLE STEPPER=stepper_z1 enable=1
	SET_STEPPER_ENABLE STEPPER=extruder enable=0
	
    DISABLE_ALL_SENSOR
    SET_GCODE_OFFSET Z=0 MOVE=0
    BED_MESH_CLEAR
    G31
    CLEAR_LAST_FILE
    BEEP I=2 DUR=500
    M18 X Y
```