frap129/OpenQ1 #28: Section 'shaketune' is not a valid config section

Megastrong1, 2024-12-07:
Hi! After installing the firmware it gives the following error, did I do something wrong or is this a feature?
[image]

deveth0, 2024-12-11:
See #20, you need to update all packages first.

frap129, 2024-12-12:
Like deveth0 said, updating packages should solve this.

Megastrong1, 2024-12-12:
This doesn't really solve the problem. Only after I commented out [shaketune] did it work. 
[image]

frap129, 2024-12-12:
In the updates section of Fluidd, did you update everything? Disabling shaketune only hides the error

Megastrong1, 2024-12-12:
I tried to restart using the "RESTART FIRMWARE" button

Megastrong1, 2024-12-12:
And also "RESTART KLIPPER"

Megastrong1, 2024-12-12:
I can demonstrate it on video. Perhaps I'm doing something wrong.

frap129, 2024-12-12:
> I tried to restart using the "RESTART FIRMWARE" button

That does not restart klipper, that restarts the MCU firmware. The only way to restart from Fluid is using the menu button in the top right, then click services, then the circle arrow next to klipper

Megastrong1, 2024-12-12:
Let me try your method and then it will be clear.

frap129, 2024-12-12:
> In the web interface, Fluidd performed all updates and Klipper still did not start. I had to copy q1.cfg and add it to the parent directory and comment it out [shaketune]

Copying q1.cfg to the parent directory doesn't do anything. I think when you restarted to apply the new config, you inadvertently restarted and solved your issue already.

Megastrong1, 2024-12-12:
[image]

In printer.cfg I changed the path

Megastrong1, 2024-12-12:
https://disk.yandex.ru/i/B9UTcPHtAmaWHg
Here's a video of what it all looks like.

frap129, 2024-12-12:
If you've updated everything and that is still happening, something must have gone wrong with the update. You can reinstall shaketune by following the instructions here: https://github.com/Frix-x/klippain-shaketune

or, your fix works fine if you dont plan on using shaketune

Megastrong1, 2024-12-12:
My fix works and that's enough for me. I just decided to immediately notify you that something is broken somewhere.

Megastrong1, 2024-12-12:
I tried reinstalling and it worked.

messschuh, 2024-12-12:
i stuck at same issue today. reinstall with `wget -O - https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh | bash` solve the problem for now.

aidan-gibson, 2025-01-29:
Confirmed also got this issue on a fresh install

frap129, 2025-01-29:
> Confirmed also got this issue on a fresh install

Yep, the base image needs to be rebuilt, but the upstream repo mine was based on was rebased, so that requires a bit of rework and testing to do.

frap129, 2025-01-29:
Leaving this open for visibility to others

closed 2026-10-23 by frap129