frap129/OpenQ1 #29: AUTO_Z_SAVE_GCODE_OFFSET

Megastrong1, 2024-12-08:
A small wish. I would like to see a macro for saving AUTO_Z_SAVE_GCODE_OFFSET in future versions. Which might look like this: 

[gcode_macro SAVE_AUTO_Z_GCODE_OFFSET]
gcode:
    AUTO_Z_SAVE_GCODE_OFFSET
    SAVE_CONFIG

Phil1988, 2025-05-11:
Does this work?

```
[gcode_macro SAVE_AUTO_Z_GCODE_OFFSET]
rename_existing: _AUTO_Z_SAVE_GCODE_OFFSET
gcode:
    _AUTO_Z_SAVE_GCODE_OFFSET
    SAVE_CONFIG
```

frap129, 2025-05-11:
> [@Megastrong1](https://github.com/Megastrong1) Does this work for you?
> 
> ```
> [gcode_macro SAVE_AUTO_Z_GCODE_OFFSET]
> rename_existing: _AUTO_Z_SAVE_GCODE_OFFSET
> gcode:
>     _AUTO_Z_SAVE_GCODE_OFFSET
>     SAVE_CONFIG
> ```

There's no need to rename it, since you aren't overriding the original name.