frap129/OpenQ1 #30: Quality of life improvements

gullygossner, 2025-01-04:
This might not be there proper method for such requests so feel free to close and point me to the proper medium to suggest such things if it isn't. 

1. I think it is beneficial to load/unload filament at the purge bucket location. I had modified my macros do so so in the stock firmware but the macros appear to be read only in this interation so that doesn't seem to be an option. 
2. This might be a very personal use case but I prefer to unload my filaments at the end of the print since I rarely print with the same filament back to back and having to manually do filament swaps is a hassle. I've operated this way on other printers for ages without issue. This would also require a load sequence or an extruder move in the start print macro to bring the filament to the nozzle. 
3. There doesn't seem to be the ability to do PID tunes on the heaters , again since the file is read only. Is it generally accepted that the pid values are consistent across machines or am I missing something?

BrokenR3C0RD, 2025-01-05:
You should be able to edit everything over SSH or through something like Filazilla or WinSCP connected to your printer. I did this myself as I had some settings to switch that required some modifications to get working (`BED_MESH_CLEAR`, skew correction, etc), it just means you won't be able to update automatically.

I was able to run PID tuning afterwards by removing the `control: pid` and other PID parameters from the heater sections of `config/q1.cfg`, then restarting Klipper and running them as usual. Results definitely vary by printer, especially the Kd values, and this also changes depending on what temperatures you're calibrating for.

gullygossner, 2025-01-05:
> You should be able to edit everything over SSH or through something like Filazilla or WinSCP connected to your printer. I did this myself as I had some settings to switch that required some [...]

What does the update process look like if one makes changes?

frap129, 2025-01-07:
> This might not be there proper method for such requests so feel free to close and point me to the proper medium to suggest such things if it isn't.
>
>     1. I think it is beneficial to load/unload [...]

To clarify the editing questions: All of the files are editable, just not through the web ui. If you you want to be able to receive configuration updates, bug fixes, etc from this repo, you should not edit anything other than `printer.cfg`.  That being said, the only thing that prevents you from doing is editing existing macros. Everything else can be overridden in printer.cfg. For example, if you want to tune PID values, just add them to your `printer.cfg` with the proper heading: 
```
[extruder]
pid_Kp=33.555
pid_Ki=4.76
pid_Kd=59.141
```

For 1, you can add a custom macro in printer.cfg. Loading/unloading over the purge area would be a nice improvement, if you could share your macro I would be happy to include it in the main config.

For 2, that is certainly a user specific thing. You can add custom gcode to your start/end gcode in your slicer to accomplish that.

gullygossner, 2025-01-07:
For point 1, I had been using the macros from the linked repo below to load/unload at three bucket. 
https://github.com/jrymk/q1pro-mods/tree/main/config

Thank you for the clarification on editing. I moved a copy of macros/Q1.cfg out of the q1 folder and commented out an include statement in order to customize. I was also happy to see that kiauh was included so that I could use mainsail over fluidd. Thank you for the work!

frap129, 2025-02-27:
I've updated the load/unload macros to be more like jrymk's. The now happen over the bucket, wipe after, and attempt tip forming.

closed 2025-02-27 by frap129