frap129/OpenQ1 #31: Bed mesh before nozzle clean causes potential bucket misses

BrokenR3C0RD, 2025-01-07:
The current start GCode does the bed mesh before cleaning the nozzle. Although this makes sense as it cleans the nozzle in preparation for the print immediately afterwards, this causes some oddities in the nozzle bucket routines.

Specifically during print start, this can cause the toolhead to be offset by a small amount during nozzle cleaning, which can result in the toolhead not entirely activating the purge bucket, or missing the felt strip entirely during printing. The nozzle should instead be cleaned before the bed mesh is calculated. Additionally, by starting the bed heating, doing the nozzle clean, then waiting for it to reach temperature before calculating the bed mesh, this can save some time since the bed takes longer to heat than the nozzle in many cases.

frap129, 2025-01-07:
I've tried doing the cleaning before meshing, the nozzle will ooze during the mesh, gunk up the bed and nozzle, and defeat the purpose of cleaning entirely, so I won't be making that change. That being said, the toolhead position shouldn't be offset after meshing. What size object were you trying to print when this occured? I haven't noticed it, but I generally print smaller parts with small mesh areas

BrokenR3C0RD, 2025-01-07:
> I've tried doing the cleaning before meshing, the nozzle will ooze during the mesh, gunk up the bed and nozzle, and defeat the purpose of cleaning entirely, so I won't be making that change.
I haven't had this personally; I'm usually printing with PETG, and have my macros set to cool down the nozzle before running the rest of the bed mesh. I mainly had to make this change on my machine since I was setting up skew correction, which needs to be applied before the bed mesh but after the nozzle clean.

> That being said, the toolhead position shouldn't be offset after meshing. What size object were you trying to print when this occured? I haven't noticed it, but I generally print smaller parts with small mesh areas
It depends on the print; some were nearly full bed, others were not. It seems strange but this has happened to me on multiple occasions, and I have varying positions on the felt pads that show the nozzle is not very aligned with the middle most of the time.

frap129, 2025-01-14:
> It depends on the print; some were nearly full bed, others were not. It seems strange but this has happened to me on multiple occasions, and I have varying positions on the felt pads that show the nozzle was not very aligned with the middle most of the time.

That is intentional, the offset is slightly randomized to help prevent wearing down the pad quickly. Its only a 4mm range so it shouldn't miss the pad entirely, perhaps the exact position varies slightly from machine to machine. If you are able to edit a file over ssh, could you try changing `range(0,40)` on this line: https://github.com/frap129/OpenQ1/blob/main/config/gcode_macro.cfg#L81 to `range(0, 30)`? This should still give some benefit of slightly randomizing the pad position, but with 1mm less variance

frap129, 2025-03-01:
The shortened range is now default in the configs

closed 2025-03-01 by frap129