frap129/OpenQ1 #34: Installation guide corrections

MicroSur, 2025-01-30:
In the installation instruction:

cp ~/OpenQ1/mcu_configs/.config.toolhead.klipper .config
It didn't work with the latest versions of Klipper, I have to fill in all the items manually.

And it is worth mentioning that before this command you need to quickly press the reset button on the toolhead board twice.
python3 scripts/flashtool.py -d /dev/ttyS2 -f ~/klipper/out/klipper.bin

Thank you for your work!

frap129, 2025-02-18:
Thank you for reporting these!

> cp ~/OpenQ1/mcu_configs/.config.toolhead.klipper .config It didn't work with the latest versions of Klipper, I have to fill in all the items manually.

This was because klipper configs have drifted, new features have been added upstream and the configs no longer applied. I have regenerated them, and they are shown correctly after copying them. Thanks for finding this

> And it is worth mentioning that before this command you need to quickly press the reset button on the toolhead board twice. python3 scripts/flashtool.py -d /dev/ttyS2 -f ~/klipper/out/klipper.bin

Did you run `systemctl stop klipper` as the install doc suggests? Quickly pressing the reset button twice should not be necessary, but it may have resolved the issue by disconnecting the toolhead from klipper temporarily, which the recommended `systemctl stop klipper` would also accomplish.

frap129, 2025-02-27:
I have updated the guide to make the `systemctl stop klipper` command part of the normal instructions, rather than as a note.

closed 2025-02-27 by frap129