frap129/OpenQ1 #35: Tool head board

aleclaz124, 2025-02-02:
Might be a hardware issue but it seems my boot and reset buttons in the tool board may just not work. They don’t seem to press and I assume pressing reset should atleast make the power led turn off. Is this correct?

frap129, 2025-02-05:
Yes, pressing reset should make the power led turn off briefly. They hard hard to press, I found it was easiest for me to press it with a small Allen wrench, like the ones the come with the printer.

aleclaz124, 2025-02-07:
Yea couldn’t get it to press no matter how hard I tried with an Allen key. I’m thinking at some point I’ll buy a new tool head board from qidi and see if that one works. I ended up reverting back to stock for now but I do appreciate your response and will likely be back for another go.

closed 2025-02-17 by frap129