frap129/OpenQ1 #37: New updates?

NpFaith, 2025-02-16:
Hi, I’m just asking if there will be new updates?
Thanks

frap129, 2025-02-18:
There may be bug fixes, compatibility improvements with future klipper/moonraker versions, and improvements to the docs. What do you mean by new updates?

zakfromevil, 2025-02-22:
Hello. How about adding HID-FLASH (stm32 flash) for update klipper on MCU-U1 without SD-card?

frap129, 2025-02-23:
> Hello. How about adding HID-FLASH (stm32 flash) for update klipper on MCU-U1 without SD-card?

I tested and considered using it in the guide, however, I think the risk to corrupting the bootloader and being unable to flash the mcu without diving into the cmdline with stm32flash was too high to suggest it to any users without experience with these devices.

Megastrong1, 2025-03-10:
Actually, the only thing missing is control via the screen. Are there any guides on how to make the interface active?