frap129/OpenQ1 #38: BTT EDDY DUO - Installed

straga, 2025-02-18:
Use for: Probe, Home Z end stop, Z-offset.

[image]

My config for 3d printer with Eddy_NG:
https://github.com/straga/3d_printer/tree/master/OpenQ1pro/config_eddy_ng/

or with what inside Klipper fot BTT EDDY
https://github.com/straga/3d_printer/tree/master/OpenQ1pro/config/

frap129, 2025-02-18:
Thank you for sharing this! I've pinned this issue so that others may find your work. I've been interested in eddy pobes myself, so I may give this a try at some point.

frap129, 2025-02-27:
I am converting this issue into a discussion, since its not really an issue.