frap129/OpenQ1 #45: What am I missing here?

danb35, 2025-03-10:
I have to be missing something major, because otherwise I'm just running into too many problems with this.  I installed it on my Q1 Pro (on its own EMMC), following the install docs, but have been running into problem after problem:

* "shaketune is not a valid configuration section" as in https://github.com/frap129/OpenQ1/issues/28, resolved by reinstalling Shaketune.
* Can't PID tune.  Well, I can, but I can't save the results using Fluidd, as they're included from `q1.cfg`.
* "Probe triggered prior to movement" as in https://github.com/frap129/qidi_auto_z_offset/issues/9.  No resolution as yet to this one.
* OK, it's a step backward, but I can manually adjust Z offset.  But once again, I can't save the new value using Fluidd, as it's also included from `q1.cfg`.
* Well, I can always SSH in and make the edits to `q1.cfg` via nano.  Restart and all is good, right?
* Nope.  When I try to print something, I'm now getting this error:
```
16:18:52 
!! Error evaluating 'gcode_macro PRINT_START:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'calibrated_z_offset'
```

It looks like this configuration puts the printer in a state where `SAVE_CONFIG` mostly doesn't work, which makes me think I have to be doing something wrong.  But darned if I can figure out what it is.  Any suggestions, or pointers on where to look?

frap129, 2025-03-10:
>     * Can't PID tune.  Well, I can, but I can't save the results using Fluidd, as they're included from `q1.cfg`.
You can manually add the PID tune values  to `printer.cfg`, just add an `[extruder]` heading and the values. This is a bug/missing feature in klipper, `SAVE_CONFIG` doesn't handle split configs well. 

Same workaround as above, but see 

>     * Well, I can always SSH in and make the edits to `q1.cfg` via nano.  Restart and all is good, right?
Like the first line in that file says, don't modify it, as it will break updating from moonraker.   Anything you want to change or add can be done in `printer.cfg`. If you don't want to get updates from this repo, you can remove the update config from `moonraker.conf` and edit them all you want.

>     * OK, it's a step backward, but I can manually adjust Z offset.  But once again, I can't save the new value using Fluidd, as it's also included from `q1.cfg`.
>     * Nope.  When I try to print something, I'm now getting this error:
> !! Error evaluating 'gcode_macro PRINT_START:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'calibrated_z_offset'

This is because PRINT_START is trying to load the value measured by auto_z_offset. You can either:
- Add the following to `printer.cfg`
```
[auto_z_offset]
calibrated_z_offset: <manually measured value>
```

Or, if you don't plan on updating the configs from this repo, remove `AUTO_Z_LOAD_OFFSET` from `PRINT_START` in `gcode_macro.cfg`, and manually add your offset.

> * "Probe triggered prior to movement" as in Can't run AUTO_Z_CALIBRATE - Error "probe triggered prior to movement!" qidi_auto_z_offset#9. No resolution as yet to this one.
This one I'm still trying to figure out, the plugin uses the same logic as the stock firmware, and the procedure works for myself and others, but some printers don't seem to like it. Hard to debug when I can't reproduce the issue.

danb35, 2025-03-16:
Thanks for the feedback, and apologies for the delay in following-up--I was out of town unexpectedly for a few days.  And yes, I understood the warning in `q1.cfg`, but wasn't aware of a different way to address it.  So to make sure I understand, I'd want to return `q1.cfg` to its original state, and then add to `printer.cfg` (after the line that includes q1.cfg) something like this:

```
[extruder]
control : pid
pid_Kp=44.527
pid_Ki=9.576
pid_Kd=51.763

[heater_bed]
control = pid
pid_Kp=63.418
pid_Ki=1.342
pid_Kd=749.125

[auto_z_offset]
calibrated_z_offset: 2.185
```
?

Would that allow the normal Z-offset microstepping and saving to happen?  Or would I also need to add a `[probe]` section to that file?

> Hard to debug when I can't reproduce the issue.

Understood.  Is there info I can provide, or tests I can run, to help out in this regard?

frap129, 2025-03-22:
Yep, you can add that chunk to `printer.cfg` and it will override what is in `q1.cfg`.  

By `normal Z-offset microstepping`, do you mean the UI buttons in Fluidd? or do you mean using the `TESTZ` command? `auto_z_offset` loads the calibrated offset using a `GCODE_OFFSET`, which is the same method the UI buttons in Fluidd use. Using the UI buttons doesn't allow for saving, which is why `AUTO_Z_SAVE_GCODE_OFFSET` exists, This command will store the currently set `GCODE_OFFSET` as the `calibrated_z_offset`, which can then be saved with `SAVE_CONFIG`

closed 2025-05-14 by danb35