frap129/OpenQ1 #46: Flashing EMMC Issue

Joyboy0395, 2025-03-10:
I've been trying to flash the image onto the EMMC, and the flashing process works fine. We've followed all the steps in the guide and made the necessary changes, but when we get to Part 2 and attempt to connect to the printer, it won't connect to the Wi-Fi. We're stuck at this point and have tried redoing the process several times, but the issue persists. Is there something we're missing that's preventing the Wi-Fi connection, so we can proceed with Part 2? We've been flashing the .img.xz file but it fails so we had to make it just a .img file and it worked but idk if thats the correct file that were supposed to be flashing onto the EMMC.

frap129, 2025-03-10:
A few questions:
- Did the same wifi network work on the stock firmware? The wifi dongle included only supports 2.4GHz networks.
- How did you make the img.xz into a .img? You would need to decompress it, which Balena Etcher should do automatically when flashing
-Does your wifi password have any special characters? These may need to be escaped when adding the wifi info

If you have one of the later hardware revisions (ethernet port on the board) you can start the printer using ethernet and setup wifi over ssh if need be.

Joyboy0395, 2025-03-10:
hello, 
1. yes the wifi im trying to connect is the one i was originally using on the stock firmware and it is 2.4
2. i made the img.xz into a .img from using 7-zip 
3. no my wifi password doesnt have any special characters, just numbers and letters 

[image]

Joyboy0395, 2025-03-12:
So I tried the ethernet to see if it would connect to the internet while it was on the starts abnormally page and it still isnt connecting

frap129, 2025-03-12:
You do have to disable wifi and enable Ethernet in armbian_first_run.txt, otherwise it will be disabled. Note that that file is only used for the first run, so you should reflash the emmc first before setting it up. How are you flashing btw? If you are using Balena Etcher, it should be able to automatically decompress the img.xz file

Joyboy0395, 2025-03-12:
I was justs thinking about redoing the flash to see if it was just prioritizing wifi over ethernet, I try to do the img.xz file but it does the normal sequence where it downloads and validates but once it finishes validated it errors saying something went wrong, if it a compressed image, please check that the archive is not corrupted

Joyboy0395, 2025-03-12:
So i tried doing it so it connected to the ethernet and it still didnt show up on angry ip scanner or my router

frap129, 2025-03-16:
Do the case lights in the printer turn on after a while? If you aren't getting anything over ethernet, but the case lights turn on, then the printer is booting properly and failing to connect to your network, which would point to a configuration issue. If the case lights do not turn on even after 10+ minutes, then it's likely that there has been an issue with your download (corrupted file) or the flashing process.

Joyboy0395, 2025-03-16:
I got it working i tried a different flashing app and it worked, I just had one question after I ssh into it and put the Microsd with klipper in and let it restart is the screen still supposed to say that it started abnormally or is that normal?

frap129, 2025-03-16:
The screen displays the started abnormally message when it can't communicate with the printer's host board. OpenQ1 doesn't have screen support, so it will keep displaying that.

closed 2025-03-16 by frap129