frap129/OpenQ1 #48: Slicer-Setup changes

fluffea, 2025-03-21:
The Slicer Setup page should be updated from:

PRINT_START BED=[first_layer_bed_temperature] HOTEND=[first_layer_bed_temperature] CHAMBER=[chamber_temperature]

to:
PRINT_START BED=[first_layer_bed_temperature] HOTEND=[nozzle_temperature_initial_layer] CHAMBER=[chamber_temperature]

Small change, but for someone who is just following the guide verbatim, it would help, otherwise you would get "!! Extrude below minimum temp" in the console because it's trying to extrude filament at whatever temperature the bed is at.

frap129, 2025-03-22:
I'm a little confused here. Hotend should be set to `first_layer_temperature`, which is what the slicer setup doc has. You say it should be updated from `HOTEND=[first_layer_bed_temperature]`, which is not that doc has. What is the suggestion here?

fluffea, 2025-03-22:
I apologize, I guess it was just my dyslexia. For some reason I read it that way. Then, it would be very helpful to have the the start gcode be cut and pasteable.

closed 2025-03-22 by fluffea

frap129, 2025-03-29:
Good point, I've added the gcode as text so it can be copy/pasted