frap129/OpenQ1 #51: New Q1 Pro not detecting wifi

Ivanuke, 2025-04-11:
Previously I set up 12 Q1 Pros with the open Q1 firmware without issue.

I purchased 6 more Q1 Pros recently, and after flashing the EMMC, making the necessary edits to the txt file, my router is not detecting the printer. I noticed these machines have a different Wifi Dongle, it is a CF-WU815N. Not sure how to fix this, none of the 6 machines are detected after flashing.

Thanks

frap129, 2025-04-15:
I did some googling to see what chipset this adapter uses, and it appears to be an RTL8188GU which should be supported. Some people reported [here](https://github.com/McMCCRU/rtl8188gu/issues/71) that the firmware for it isn't installed in some cases.

Is it possible for you to set one of the printers up with ethernet? If you can try to install the firmware for the wifi dongle with 
```
sudo wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/rtlwifi/rtl8710bufw_UMC.bin -O /lib/firmware/rtlwifi/rtl8710bufw_UMC.bin
sudo wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/rtlwifi/rtl8710bufw_SMIC.bin -O /lib/firmware/rtlwifi/rtl8710bufw_SMIC.bin
sudo modprobe -r rtl8xxxu && sudo modprobe rtl8xxxu
```
then try to connect to wifi with `nmtui`, that would be helpful.

I only ask because the original wifi dongle QIDI used was also *allegedly* an RTL8188GU, but was actually an AIC8800. If this firmware is all that is needed, I can rebuild the image with that firmware pre-installed.

Ivanuke, 2025-04-15:
Sorry I forgot to come back and update. I did have to connect via ethernet, and then I installed rtl8710bufw_SMIC.bin

I then used nmtui as you suggested to connect to wifi and it now works!

DenTechs, 2025-07-11:
Just ran into the same issue myself, connecting with ethernet and then installing the drivers manually worked.