frap129/OpenQ1 #54: MCU U_1 Connection lost

Chriz-123, 2025-05-10:
Have followd the steps from the Tutotial until the last step from Part4 where i encountered an error, 

```
Attempting to connect to bootloader
ERROR:root:Flash Error
Traceback (most recent call last):
  File "/home/mks/katapult/scripts/flashtool.py", line 554, in run
    await flasher.connect_btl()
  File "/home/mks/katapult/scripts/flashtool.py", line 90, in connect_btl
    ret = await self.send_command('CONNECT')
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mks/katapult/scripts/flashtool.py", line 196, in send_command
    raise FlashCanError("Error sending command [%s] to Can Device"
FlashCanError: Error sending command [CONNECT] to Can Device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/mks/katapult/scripts/flashtool.py", line 633, in main
    loop.run_until_complete(sock.run(args.device, args.baud, fpath))
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/mks/katapult/scripts/flashtool.py", line 561, in run
    await flasher.finish()
  File "/home/mks/katapult/scripts/flashtool.py", line 275, in finish
    await self.send_command("COMPLETE")
  File "/home/mks/katapult/scripts/flashtool.py", line 196, in send_command
    raise FlashCanError("Error sending command [%s] to Can Device"
FlashCanError: Error sending command [COMPLETE] to Can Device

```

After that i tried it again from scratch,but same result
Also cannot revert to Stockfirmware.

Always getting mcu 'mcu': Unable to connect fluidd ui

frap129, 2025-05-10:
Have you verified that the toolhead is in bootloader mode when flashing? The buttons are tricky to get right. Run "lsusb" over SSH and note the result, then try to put it the toolhead into bootloader mode and run "lsusb" again. If it has been done correctly, there should be a new raspberry pi/rp2040 device listed.

Chriz-123, 2025-05-10:
Yes, was able to put the katapult file before.
When pressing boot+reset 

Bus 004 Device 004: ID 2e8a:0003 Raspberry Pi RP2 Boot

appears under lsusb

frap129, 2025-05-11:
I'm not.sure why it wouldn't work a second time, but that doesn't seem to be this main problem. The 'mcu' unable to connect is referring to the mainboard MCU, did you follow the steps to flash that with a micro sdcard?

Chriz-123, 2025-05-11:
Yes i did, and every time i put in a sd card with a qd_mcu.bin ill get a gd_mcu.CUR back

frap129, 2025-05-11:
I doubt that it's damaged, nothing about the flashing procedure is inherently dangerous. Could you downlowd the klippy.log from fluidd and upload it here? That should be under the system tab in fluidd

Chriz-123, 2025-05-11:
here a link to the [log](https://mega.nz/file/qxBF2TBR#_rMNErql1wcFCgDRrDYnKUUSVQNdSpRlsd7PQl24IfA)

Phil1988, 2025-05-12:
@Chriz-123 The problem you see is coming from the onboard MCU.
Replacing the toolhead cable wont affect that.

I am confused: Is OpenQ1 really using an old v0.10.0-530-g3387a9c2-dirty klipper?
If not, then you may need to upgrade your klipper (host) version.
If yes, you can try to use an old pre-compiled klipper from me just to double check:
https://github.com/Phil1988/FreeDi/blob/master/old_mainboard_and_toolhead_firmwares/v0.10.0-530_old_stock/X_4.bin
(make sure to rename it to qd_mcu.bin when placing it onto the sd card)

Phil1988, 2025-05-12:
> v0.10.0-530-g3387a9c2-dirty klipper is from Qidi Stock Image

You said you follwed the guide, but "Part 1: Installing the Armbian image" so I assumed you were using the OpenQ1 image.
You need matching klipper versions of the host and the mainboard MCU otherwise that error is normal.

So katapult was installed without issues?

Chriz-123, 2025-05-12:
> You said you follwed the guide, but "Part 1: Installing the Armbian image" so I assumed you were using the OpenQ1 image. You need matching klipper versions of the host and the mainboard MCU [...]

sitting here many hours and tryin to get it running, it was an attempt to flash back to stock. were i posted the klippy.log sorry

here is the one with openq1 [https://mega.nz/file/r4Rw1JhR#FOW4KYj833CLVOAX47WpqzoEgRKdYDyk1H4TY5jccfY](url)

Yes katapult ist installed without issue

Phil1988, 2025-05-12:
can you show the full output after
`python3 /home/mks/katapult/scripts/flashtool.py -d /dev/ttyS2 -f /home/mks/klipper/out/klipper.bin` ?

Chriz-123, 2025-05-12:
> [@Chriz-123](https://github.com/Chriz-123) The problem you see is coming from the onboard MCU. Replacing the toolhead cable wont affect that.
>
> I am confused: Is OpenQ1 really using an old [...]

When i use the X_4.bin the blue LED under the eMMC Card stays on and i dont have a connection to the printer.

> can you show the full output after `python3 /home/mks/katapult/scripts/flashtool.py -d /dev/ttyS2 -f /home/mks/klipper/out/klipper.bin` ?

give me a few minutes

Chriz-123, 2025-05-12:
mks@qidi-q1:~/katapult$ python3 scripts/flashtool.py -d /dev/ttyS2 -f ~/klipper/out/klipper.bin
Attempting to connect to bootloader
ERROR:root:Flash Error
Traceback (most recent call last):
  File "/home/mks/katapult/scripts/flashtool.py", line 554, in run
    await flasher.connect_btl()
  File "/home/mks/katapult/scripts/flashtool.py", line 90, in connect_btl
    ret = await self.send_command('CONNECT')
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mks/katapult/scripts/flashtool.py", line 196, in send_command
    raise FlashCanError("Error sending command [%s] to Can Device"
FlashCanError: Error sending command [CONNECT] to Can Device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/mks/katapult/scripts/flashtool.py", line 633, in main
    loop.run_until_complete(sock.run(args.device, args.baud, fpath))
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/mks/katapult/scripts/flashtool.py", line 561, in run
    await flasher.finish()
  File "/home/mks/katapult/scripts/flashtool.py", line 275, in finish
    await self.send_command("COMPLETE")
  File "/home/mks/katapult/scripts/flashtool.py", line 196, in send_command
    raise FlashCanError("Error sending command [%s] to Can Device"
FlashCanError: Error sending command [COMPLETE] to Can Device

frap129, 2025-05-12:
To clarify, you reattached the stock cable before running that command, yes? Katapult is flashed over USB, but Katapult is used to flash klipper over the stock serial connection after that.

Chriz-123, 2025-05-12:
nope used usb... all the time :\
but also with the stock cable ill get the same error

Chriz-123, 2025-05-12:
i think ill have damaged the uart cable or the toolhead, but ill get a connection with usb, to the toolhead.
Anyway i have orderd both now, new toolhead pcb and cable

Phil1988, 2025-05-12:
@frap129 I dont think it is needed, but do one have to click/double click the reset button to initiate the boot mode before using that command?

@Chriz-123 can you show your klipper version and recomile it just to be sure?
``
cd ~/klipper
make menuconfig
```
take a screenshot, then:
```
make clean
make -j4
```
and retry
```
python3 /home/mks/katapult/scripts/flashtool.py -d /dev/ttyS2 -f /home/mks/klipper/out/klipper.bin
```

Chriz-123, 2025-05-12:
[Screenshot](https://mega.nz/file/HkZGwCSI#5D5tzsOSftOWAMBZ3WggnvOazcqE4mjKx_Aut3WwYCM)
```
mks@qidi-q1:~/klipper$ make clean
make -j4
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/spi_software.o
  Compiling out/src/i2c_software.o
  Compiling out/src/thermocouple.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/sensor_angle.o
  Compiling out/src/sensor_mpu9250.o
  Compiling out/src/sensor_lis2dw.o
  Compiling out/src/sensor_ldc1612.o
  Compiling out/src/sensor_bulk.o
  Compiling out/src/rp2040/main.o
  Compiling out/src/rp2040/watchdog.o
  Compiling out/src/rp2040/gpio.o
  Compiling out/src/rp2040/adc.o
  Compiling out/src/rp2040/timer.o
  Compiling out/src/rp2040/bootrom.o
  Compiling out/src/generic/armcm_boot.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_reset.o
  Compiling out/src/generic/timer_irq.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/rp2040/serial.o
  Compiling out/src/generic/serial_irq.o
  Compiling out/src/rp2040/hard_pwm.o
  Compiling out/src/rp2040/spi.o
  Compiling out/src/rp2040/i2c.o
  Preprocessing out/src/rp2040/rp2040_link.ld
  Building out/compile_time_request.o
Version: v0.12.0-242-g11f04ba1b
  Linking out/klipper.elf
  Creating bin file out/klipper.bin
mks@qidi-q1:~/klipper$
```

mks@qidi-q1:~/klipper$ python3 /home/mks/katapult/scripts/flashtool.py -d /dev/ttyS2 -f /home/mks/klipper/out/klipper.bin
Attempting to connect to bootloader
ERROR:root:Flash Error
Traceback (most recent call last):
  File "/home/mks/katapult/scripts/flashtool.py", line 554, in run
    await flasher.connect_btl()
  File "/home/mks/katapult/scripts/flashtool.py", line 90, in connect_btl
    ret = await self.send_command('CONNECT')
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mks/katapult/scripts/flashtool.py", line 196, in send_command
    raise FlashCanError("Error sending command [%s] to Can Device"
FlashCanError: Error sending command [CONNECT] to Can Device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/mks/katapult/scripts/flashtool.py", line 633, in main
    loop.run_until_complete(sock.run(args.device, args.baud, fpath))
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/mks/katapult/scripts/flashtool.py", line 561, in run
    await flasher.finish()
  File "/home/mks/katapult/scripts/flashtool.py", line 275, in finish
    await self.send_command("COMPLETE")
  File "/home/mks/katapult/scripts/flashtool.py", line 196, in send_command
    raise FlashCanError("Error sending command [%s] to Can Device"
FlashCanError: Error sending command [COMPLETE] to Can Device

frap129, 2025-05-12:
> [@frap129](https://github.com/frap129) I dont think it is needed, but do one have to click/double click the reset button to initiate the boot mode before using that command?

No, it explicitly will not work if you are in boot mode. The toolhead should be in normal operating mode.

@Chriz-123 make sure to run `sudo systemctl stop klipper` before running the katapult command

Chriz-123, 2025-05-13:
seems that the toolhead was toasted, after installing new toolhead board 

ill get this on fluidd ui

```
MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.12.0-242-g11f04ba1b-dirty
MCU(s) which should be updated:
mcu: Current version v0.10.0-530-g3387a9c2-dirty-20221222_010328-mkspi
U_1: Current version v0.12.0-242-g11f04ba1b
Up-to-date MCU(s):

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'mcu': Command format mismatch: query_adxl345 oid=%c rest_ticks=%u vs query_adxl345 oid=%c clock=%u rest_ticks=%u
```
👍