frap129/OpenQ1 #56: SAVE_CONFIG

jStrahan73, 2025-06-21:
Why do i get this error when i try to save my pid tune values

SAVE_CONFIG section 'extruder' option 'control' conflicts with included value

frap129, 2025-07-08:
SAVE_CONFIG doesn't play nicely with multi-file configs, this is a known klipper issue. You can open your printer.cfg manually and add an [extruder] header followed by your PID values

AlexandrTyshchenko, 2026-10-03:
This isn't really a Klipper bug — it's intentional: `SAVE_CONFIG` refuses to autosave options that are already set in an *included* file (`configfile._disallow_include_conflicts`). Options set in the main `printer.cfg` are fine; Klipper just comments them out when saving. `q1.cfg` (included) defines `control`/`pid_Kp`/`pid_Ki`/`pid_Kd` for `[extruder]`, `[heater_bed]` and `[heater_generic chamber]`, so `PID_CALIBRATE` + `SAVE_CONFIG` can never work out of the box.

A possible fix: move those PID lines from `q1.cfg` into `config/printer.cfg` (under `[extruder]`, `[heater_bed]`, `[heater_generic chamber]` headers). The catch is migration: `printer.cfg` is only copied at install time, while `q1.cfg` updates through Moonraker, so existing users would get "Option 'control' in section 'extruder' must be specified" until they add the lines to their `printer.cfg` (a release note with a copy-paste block would cover it).

Happy to send a PR if you'd like to go that way.